phasorpy/phasorpy #6: Add a lifetime calculator module

cgohlke, 2023-10-01:
This issue proposes the addition of a `phasorpy.lifetime` module for fluorescence lifetime and energy transfer (FRET) calculations with the following features:

- calculate phasor coordinates from time- and frequency-domain signals
- convert between
    - phasor coordinates
    - phase and modulation
    - apparent single lifetimes
    - lifetime mixtures with fractional intensities or pre-exponential amplitudes
- reference phasor coordinates
- generate synthetic signals, time domain decay and frequency domain wave
- calculate FRET as a function of 
    - donor and acceptor lifetime
    - FRET efficiency
    - donors with FRET
    - donor in acceptor channel
    - direct acceptor excitation
    - donor and acceptor channel background
    - frequency
- n-dimensional aware

Implementations of exponential or global fitting are not planned.

Some of the implementation can be taken from the [vLFD Lifetime Demo](https://www.cgohlke.com/vlfd_lifetime_demo.png) (written in ~2007). That requires the following steps:
- port code from Python 2 to 3
- replace C extension module with numpy or Cython
- make n-dimensional aware
- add tests
- add documentation
- convert interactive app to tutorial or console script

Mariochem92, 2023-10-28:
Hello everyone,

I'd like to share a set of Python functions in this code snippet for the proposed `phasorpy.lifetime`  module, with the hope that they can be of assistance in addressing some of the features described in the project request.

The code relies on a few external libraries: 
- **NumPy**: Used for various mathematical operations, including array handling and calculations.

- **Shapely**: Required for working with polygons and geometric operations. Install it using  `pip install shapely`.

- **Matplotlib**: Utilized for generating plots and visualizing data. Install it using `pip install matplotlib`.

Here's an overview of the key functions and their roles in the context of your project:

1. `phasor_coordinates_FD(phi, mod)`:
   - Purpose: Converts phase and modulation data into phasor coordinates in the frequency domain.
   
2. `ph_coordinates2phmd(g, s)`:
   - Purpose: Converts phasor coordinates back into phase and module.

3. `polished_hist(g, s, hist_bins)`:
   - Purpose: Generates a polished 2D histogram from provided phasor data.

4. `contour_plot(histogram, levels, limits)`:
   - Purpose: Creates a contour plot from a histogram with specified contour levels and limits.

5. `analyze_contourplot(contour_segments)`:
   - Purpose: Analyzes contour segments of the phasor plot histogram and identifies both phasors and regions of interest.

6. `simple_lt(g, s, flim_frequency)`:
   - Purpose: Calculates lifetime values from phasor coordinates and FLIM frequency, as τ<sub>m</sub> and τ<sub> φ</sub>
   
7. `lifetime_computation(g, s, flim_frequency)`:
   - Purpose: Computes lifetime values and standard deviations from phasor coordinates and FLIM frequency with error propagation

8. `refplot(names, lifetimes, flim_frequency)`:
   - Purpose: Generates reference phasor coordinates based on sample lifetimes and FLIM frequency.

9. `Segment_IntensityFractions(RefA, RefB, target, labels)`:
   - Purpose: Calculates intensity fractions between two reference points and a target phasor.

I've left the FRET-related functionalities as they are, as I'm not experienced in that domain. If you have any specific guidance or if you'd like to explore an object-oriented approach, please let me know.

Warm regards,
Mario

--------------------

```
import numpy as np
from shapely.geometry import Polygon
import matplotlib.pyplot as plt

def phasor_coordinates_FD(phi, mod):
# Convert phase and module information into phasor coordinates in the frequency domain.
# Parameters:
# - phi: Phase angle in radians.
# - mod: Module.
# Returns:
# - g: Real part of the phasor coordinates.
# - s: Imaginary part of the phasor coordinates.
#
# This function takes phase and moduleinformation (phi and mod) and computes the corresponding phasor
# coordinates (g, s) in the phasor plot. It is valuable for converting phase and modulation data
# into phasor representations.
  g = mod * np.cos(phi)
  s = mod * np.sin(phi)
  return g, s

def ph_coordinates2phmd(g, s):
# Convert phasor coordinates into phase and module
[... 181 lines cut ...]
    M = 1 / np.sqrt(1 + (omega * tau)**2)
    phi = np.arctan(omega * tau)
    g_ref.append(M * np.cos(phi))
    s_ref.append(M * np.sin(phi))
  return g_ref, s_ref

def Segment_IntensityFractions(RefA, RefB, target, labels):
    # Calculate intensity fractions between two reference points and a target point.
    # Parameters:
    # - RefA: Coordinates of the first reference point.
    # - RefB: Coordinates of the second reference point.
    # - target: Coordinates of the target phasor.
    # - labels: A dictionary for labeling categories.
    # Returns a list of intensity fractions.

    Tot_distance = ((RefA[0] - RefB[0])**2 + (RefA[1] - RefB[1])**2)**0.5
    distanceA = ((RefA[0] - target[0])**2 + (RefA[1] - target[1])**2)**0.5
    distanceB = ((RefB[0] - target[0])**2 + (RefB[1] - target[1])**2)**0.5

    fractionA = distanceB / (distanceA + distanceB)
    fractionB = distanceA / (distanceA + distanceB)

    ratios = [fractionA, fractionB]
    return ratios
```

cgohlke, 2023-10-28:
Thanks for the input and code. That functionality will certainly be integrated in this library. Histogramming, analysis of histograms/distributions, and plotting will be separated into modules. 

Numpy and matplotlib are already core requirements. Rather than working directly with shapes, we might want to work via masks.

I'm currently focusing on I/O, #10, #11, and other formats. It will take a while before I can start working on this issue.

Mariochem92, 2023-10-28:
Thank you for the update! I appreciate your consideration of the input and code.

Regarding the use of masks, it sounds like a useful approach, especially for handling images. In this case I try to handle segments of the contour plot on the phasor cloud. I know it's customary to go with normal distribution approaches, but I had in mind a fully non-parametric contour plot in 2D. In case you suggest to mask the phasor plot that actually sounds interesting!

Separating the histogramming, analysis of histograms/distributions, and plotting into modules sounds like a great idea for better organization and modularity.

Please take your time to address these priorities, and I'll be available to eventually discuss when the time is right.

cgohlke, 2024-08-06:
All proposed features have been implemented.

closed 2024-08-06 by cgohlke

cgohlke, 2024-08-06:
> I'd like to share a set of Python functions in this code snippet

Reformatted with black for better readability:
```python
import matplotlib.pyplot as plt
import numpy as np
from shapely.geometry import Polygon

def phasor_coordinates_FD(phi, mod):
    """
    Convert phase and module information into phasor coordinates in the
    frequency domain.
    Parameters:
    - phi: Phase angle in radians.
    - mod: Module.
    Returns:
    - g: Real part of the phasor coordinates.
    - s: Imaginary part of the phasor coordinates.

    This function takes phase and moduleinformation (phi and mod) and
    computes the corresponding phasor
    coordinates (g, s) in the phasor plot. It is valuable for converting
    phase and modulation data
    into phasor representations.

    """
    g = mod * np.cos(phi)
    s = mod * np.sin(phi)
    return g, s

def ph_coordinates2phmd(g, s):
    """
    Convert phasor coordinates into phase and module
    Parameters:
    - g: Real part of the phasor coordinates.
    - s: Imaginary part of the phasor coordinates.
    Returns:
    - phi: Phase angle in radians.
    - mod: Module.

    This function takes phasor coordinates (g, s) and computes the
    corresponding phase angle (phi) in radians
    and module (mod). It is useful for transforming phasor data into
    more interpretable phase and
    module information for further analysis or visualization.

    """
    phi = np.arctan2(s, g)
    mod = np.sqrt(g**2 + s**2)
    return phi, mod

def polished_hist(g, s, hist_bins):
    """
    Generate a 2D histogram from provided phasor data.
    Parameters:
    - g: Real part of data points.
    - s: Imaginary part of data points.
    - hist_bins: Number of bins for histogram generation.
    Returns:
    - hist: The 2D histogram data.
    - filtered_array: A binary mask indicating valid data points.
    - limits: Extent limits defining the data range.
    - hist_color: Histogram values at specific data points.

    This function creates a polished 2D histogram using provided data
    points, with a specified number of bins.
    It returns the histogram data, a binary mask indicating valid data
    points, extent limits defining the data range, and histogram values
    at specific data points. This polished histogram is useful for
    visualizing data distribution and further analysis.
[... 206 lines cut ...]
    Calculate intensity fractions between two reference points and a
    target point.
    Parameters:
    - RefA: Coordinates of the first reference point.
    - RefB: Coordinates of the second reference point.
    - target: Coordinates of the target phasor.
    - labels: A dictionary for labeling categories.
    Returns a list of intensity fractions.

    """
    Tot_distance = ((RefA[0] - RefB[0]) ** 2 + (RefA[1] - RefB[1]) ** 2) ** 0.5
    distanceA = (
        (RefA[0] - target[0]) ** 2 + (RefA[1] - target[1]) ** 2
    ) ** 0.5
    distanceB = (
        (RefB[0] - target[0]) ** 2 + (RefB[1] - target[1]) ** 2
    ) ** 0.5

    fractionA = distanceB / (distanceA + distanceB)
    fractionB = distanceA / (distanceA + distanceB)

    ratios = [fractionA, fractionB]
    return ratios

```

cgohlke, 2024-08-06:
Hi [Mariochem92](https://github.com/Mariochem92),

thanks again for sharing your code. We have now implemented much of your proposed functionality. If I understood your code correctly, these are the direct replacements of your functions in the PhasorPy library:

- `phasor_coordinates_FD`: [`phasor_from_polar`](https://www.phasorpy.org/stable/api/phasor.html#phasorpy.phasor.phasor_from_polar)
- `ph_coordinates2phmd`: [`phasor_to_polar`](https://www.phasorpy.org/stable/api/phasor.html#phasorpy.phasor.phasor_to_polar)
- `simple_lt`: [`phasor_to_apparent_lifetime`](https://www.phasorpy.org/stable/api/phasor.html#phasorpy.phasor.phasor_to_apparent_lifetime)
- `lifetime_computation`: use `phasor_to_apparent_lifetime`, then numpy/scipy for the statistics
- `refplot` -> [`phasor_from_lifetime`](https://www.phasorpy.org/stable/api/phasor.html#phasorpy.phasor.phasor_from_lifetime)
- `Segment_IntensityFractions`: [`two_fractions_from_phasor`](https://www.phasorpy.org/stable/api/components.html#phasorpy.components.two_fractions_from_phasor)

We do not yet have replacements for `polished_hist`, `contour_plot`, `analyze_contourplot`.

@bruno-pannunzio : do you have any additional comments?

bruno-pannunzio, 2024-08-06:
I agree @cgohlke that the listed functions already implemented replace the proposed functionality in @Mariochem92 code. 

> We do not yet have replacements for `polished_hist`, `contour_plot`, `analyze_contourplot`.

I agree that I don't think we have this features available. It would be very interesting to have though. I would like to see the use cases of those functions as they can be good additions.

Also, I think #13 is already covered in the library and can be closed, as the  functions implemented there are a subset of the ones proposed here.