phasorpy/phasorpy #10: Support PicoQuant PT3 and PTU files

cgohlke, 2023-10-16:
Dr Evangelos Sisamakis, Product Manager Microscopy at [PicoQuant GmbH](https://www.picoquant.com), kindly provided some resources, which may facilitate implementing a file reader for the PicoQuant PT3 and PTU file formats.

The [demo workspace dataset](https://www.picoquant.com/dl_software/SPT64/Demo_workspace_SPT64.zip) contains data files used in the [SymPhoTime Analysis tutorials](https://www.tcspc.com/doku.php/howto:start).

The [SPT 64 tutorials dataset](https://figshare.com/s/4957fcfa684daef86c23) contains data files for the [Lifetime-Fitting Using the FLIM Analysis](https://www.tcspc.com/doku.php/howto:lifetime-fitting_using_the_flim_script) tutorial.

The [Luminosa software](https://www.picoquant.com/dl_software/Luminosa/LuminosaCompactInstaller.zip) can be used freely to open data files taken with a Luminosa system.

A few Python scripts (but no ready to use library) are available to read PTU and PT3 files:

- [Python/Read_PTU.py](https://github.com/PicoQuant/PicoQuant-Time-Tagged-File-Format-Demos/blob/master/PTU/Python/Read_PTU.py): official reference implementation

- [readPTU_FLIM.py](https://github.com/SumeetRohilla/readPTU_FLIM/blob/master/readPTU_FLIM.py)

- [picoquantio](https://github.com/tsbischof/picoquantio): unreleased Python package

- [snAPI wrapper](https://github.com/PicoQuant/snAPI): for time traces and correlation curves, not images

The data T3-mode formats are described at https://www.tcspc.com/doku.php/glossary:t3-mode.

cgohlke, 2023-11-02:
I [released](https://pypi.org/project/ptufile/) a first version of the [ptufile](https://github.com/cgohlke/ptufile) library.

Mariochem92, 2023-11-05:
> I [released](https://pypi.org/project/ptufile/) a first version of the [ptufile](https://github.com/cgohlke/ptufile) library.

Thank you very much @cgohlke I tested this and it works like a charm. Do you think it would be possible to make it straightforward to export the decay curves from each pixel?

cgohlke, 2024-01-14:
Closing this issue. A reader for PTU files was added in #25. Support for PT3 files would need to be added to the ptufile library first, which seems unlikely anytime soon given that the format is deprecated for ~10 years and no formal format specification is available.

closed 2024-01-14 by cgohlke

cgohlke, 2024-12-29:
As of version 2024.12.28, the ptufile library can write TCSPC histograms to T3 image mode PTU files.