phasorpy/phasorpy #26: Tutorial for pahsorpy?

baoyi-A, 2024-01-12:
Is there a tutorial of how to use the phasorpy? like script of how to read in and process the tif stack or ptu, etc.? thaks in advance!

cgohlke, 2024-01-12:
Hi @baoyi-A. The PhasorPy library is still in its early stages of development. There are no tutorials yet but we plan to make some available at https://www.phasorpy.org/stable/tutorials/. In the meantime, check if any [other software](https://www.phasorpy.org/stable/phasor_approach.html#software) suits your needs.

schutyb, 2024-01-17:
Hi @baoyi-A adding to what @cgohlke said, for spectral phasor you can check I repository I did, it´s independet from phasorpy. [https://github.com/schutyb/hsipy](url)

cgohlke, 2024-04-24:
Several [tutorials](https://www.phasorpy.org/stable/tutorials/) are now available and are updated as new functionality is added.

closed 2024-04-24 by cgohlke