phasorpy/phasorpy #68: The function `phasor_calibrate` can't handle multiple harmonic calibration

bruno-pannunzio, 2024-05-14:
The function `phasor_calibrate` from the `phasor` module can't handle multiple harmonic calibration at the same time. Although the `skip_axes` argument is in place to be used as an indicator of the harmonic content, the broadcasting is not done correctly, particularly by the function `phasor_transform` of the `phasor` module which is called by `phasor_calibrate`.

cgohlke, 2024-05-14:
>  the broadcasting is not done correctly, particularly by the function `phasor_transform`

I don't follow. `phasor_transform` is a numpy ufunc and broadcasts according to standard numpy rules. No? I don't think it needs any fixing.

bruno-pannunzio, 2024-05-14:
I tried the following pipeline:

```
from phasorpy.io import read_fbd
from phasorpy.phasor import phasor_from_signal, phasor_calibrate, phasor_center

data_signal = read_fbd(data_path, frame=-1,channel=0, keepdims=False)
calibration_signal = read_fbd(calibration_path, frame=-1,channel=0, keepdims=False)
mean, real, imag = phasor_from_signal(data_signal, harmonic=[1,2])
mean_calib, real_calib, imag_calib = phasor_from_signal(calibration_signal, harmonic=[1,2])
reference_real, reference_imag = phasor_center(real_calib, imag_calib, skip_axes=0)
real, imag = phasor_calibrate(real, imag, reference_real, reference_imag, frequency=80, lifetime=4, skip_axes=0)
```

and I am getting an error of broadcasting with `phasor_transform`:

```
--> return _phasor_transform(real, imag, phase_zero, modulation_zero, **kwargs)

ValueError: operands could not be broadcast together with shapes (2,256,256) (2,256,256) (2,) (2,)
```

I tried expanding the dimensions of `phase_zero`and `modulation_zero`accordingly before calling `phasor_transform`and it works, so I guess we can fix the issue easily.

cgohlke, 2024-05-17:
Can this be closed now?

closed 2024-05-17 by bruno-pannunzio

cgohlke, 2024-05-17:
Fixed by #69.