phasorpy/phasorpy #99: Add methods for filtering

bruno-pannunzio, 2024-08-13:
This issue proposes the implementation of new methods for the `phasor_filter` function in the `phasor` module. For now, the only method implemented is the median, which is just a convenience wrapper for `scipy.ndimage.median_filter`. Other methods proposed (but not restricted to) are:

- The Wavelet filter, first described for FLIM in DOI: [10.1088/2050-6120/aa72ab](https://iopscience.iop.org/article/10.1088/2050-6120/aa72ab) , and also in DOI: [10.1364/BOE.420953](https://doi.org/10.1364/BOE.420953).
  The Wavelet filter was already implemented in python in [pawFLIM](https://github.com/maurosilber/pawflim).

- Future implementation of the median filter in Cython can be considered, specially for consistent handling of `NaN`.

- Although less commonly used with phasors, the mean filter and gaussian filter are also common in image processing.

cgohlke, 2024-09-27:
Hi @bruno-pannunzio : it looks like some of these filters and transforms are supposed to be applied to the un-normalized intensity and Fourier components? Should we add a `normalize` parameter to the `phasor_from_signal` function?

bruno-pannunzio, 2024-09-30:
Hi @cgohlke. Optimizing the calculations for these filters could be a valuable enhancement with being able to have the un-normalized intensity. If it doesn’t seem too complex, we could consider implementing it once higher-priority tasks are completed. Otherwise, we can defer it until it's necessary during the implementation of these methods.

cgohlke, 2024-10-04:
Anscombe transformation functions are added in #139

maurosilber, 2024-10-18:
Hi,

I'm the author of pawFLIM. Let me know if I can help somehow to include it here.

As it is explained in the README, it needs the number of photons (0-th harmonic), and the first two Fourier harmonics (or the N-th and 2N-th harmonics). It expects a complex array where the harmonics are in the first dimension: `(harmonics, *shape)`.

It uses the "un-normalized" phasors since (for TCSPC) the weighted average of phasors is just adding (binning?) the un-normalized ones before dividing by the total photon number. In math:

$$ R_i = N_i r_i $$

where

- $R$ is the Fourier coefficient,
- $N$ the photon number and
- $r$ the phasor (for the $i$-th pixel)

Then:

$$ \text{WeightedAverage}(r_1, r_2) = \frac{R_1 + R_2}{N_1 + N_2} $$

Related to this, I was looking at the [calibration section of the tutorial](https://www.phasorpy.org/docs/stable/tutorials/phasorpy_introduction/#calibrate-phasor-coordinates) and it seems that the calibration function uses the (non-weighted) phasor average of reference signal:
https://github.com/phasorpy/phasorpy/blob/a02edecf9944791ce9651a13e49778062e63da2c/src/phasorpy/phasor.py#L1141-L1143

Instead of adding an extra variable for the number of photons, maybe you could represent a phasor dataset with something like:

```python
import numpy as np
from numpy.typing import NDArray

class Dataset:
    fourier_coeffs: NDArray[np.complex128]  # shape: (harmonics, *etc)

    def phasor(self, harmonic: int | slice | None = None, /):
        if harmonic is None:
            return self.fourier_coeffs / self.fourier_coeffs[0]
        return self.fourier_coeffs[harmonic] / self.fourier_coeffs[0]
```

which would also simplify the signature of many functions that expect:
- `(real, imag)`,
- `(mean, real, imag)`,
- `(mean, real_1, imag_1, real_2, imag_2)` (this would be needed for pawFLIM)
while also encapsulating the dataset and making it more robusts to bugs such as:
- mixing the real part of one dataset with the imaginary part of another,
- passing the arguments in the wrong order `(imag, real)`

cgohlke, 2024-10-18:
Hi @maurosilber,

thanks for chiming in. Yes, we are interested in integrating your de-noising and binning methods into this library. As those need access to the un-normalized coefficients and number of photons, we shall make normalization in `phasor_from_signal` optional and also add options to pass the number of samples to some functions to do correct weighting.

> maybe you could represent a phasor dataset with something like:

I'm afraid that ship has sailed. I was actually in favor of representing phasor coordinates as complex numbers. Ultimately we decided at the start of the project to separate the real and imaginary components for several reasons:

1. All existing source code handling "phasors" known to us (back to the 1990s) use that representation.
2. Handling of complex numbers on a low level in C or Cython used to be anything but pleasant (platform and compiler dependent). That has improved much. 
3. Representing the 0th harmonic as a complex number is not necessary and wasteful.
4. To keep "symmetry" with the representations of phase/modulation and apparent single lifetime from phase/modulation.
5. Many image processing and visualization libraries and software are not aware of complex numbers.
6. Several libraries and software were not able to read complex numbers from, for example, OME-TIFF.

We also decided not to define a custom class to represent phasor coordinates but instead use numpy arrays and a procedural interface for this library.

lmalacrida, 2024-11-04:
Hi @maurosilber, Thanks for joining the PhasorPy community. We are pleased that you can contribute with the pawFLIM into PhasorPy. I think adding your filter will be a great addition to PhasorPy's existing capabilities. Please let us know if you need any help setting up this module in PhasorPy. We are looking for a community development project, and I like the idea you can implement. Please let us know how we can help you with this.

cgohlke, 2024-12-02:
Hi @bruno-pannunzio : I am working on a PR to add a `normalize` parameter to the `phasor_from_signal` function. In addition, I think we might want to add the `mean` signal intensity to the parameter lists of the `phasor_center` and `phasor_filter` functions (a breaking change). That way users won't need to keep track of the de/normalization status (the workflow would stay the same) and the filter/center functions can directly use the intensity for correct averaging or other purposes as needed (or simply ignore it, like the median functions). At that point, we might also want to return `mean` intensities from the `phasor_center` and `phasor_filter` functions (whether processed or passed through), which would also make the functions more symmetric and similar to `phasor_threshold` for example.

The `phasor_center` function would become:
```Python
def phasor_center(
    mean: ArrayLike,
    real: ArrayLike,
    imag: ArrayLike,
    /,
    *,
    skip_axis: int | Sequence[int] | None = None,
    method: Literal['mean', 'median'] = 'mean',
    **kwargs: Any,
) -> tuple[NDArray[Any], NDArray[Any], NDArray[Any]]:
```

bruno-pannunzio, 2024-12-03:
Hi @cgohlke! This is a great idea! Thanks for working on the normalization parameter, it will be needed for the wavelet filter so it's good to have that already since I will start working on that as soon as we finish the workshop next week. 

I think adding the mean to `phasor_center` and `phasor_filter` is a good idea, also to simplify some of the workflow slightly since we can store everything in a `phasor` variable. Something like this:

```Python

phasor = phasor_from_signal(signal)
reference_phasor = phasor_from_signal(reference_signal)
phasor = phasor_calibrate(*phasor, *reference_phasor, frequency, lifetime)
phasor = phasor_filter(*phasor)
phasor = phasor_threshold(*phasor, 1)

```

This could also be done now with slicing the `phasor` tuple to omit the `mean`, but this would be even simpler. Also should `phasor_calibrate` also receive the `mean` considering it uses `phasor_center`?

cgohlke, 2024-12-03:
> Also should phasor_calibrate also receive the mean considering it uses phasor_center?

Yes, `phasor_calibrate` needs `reference_mean`, but not `mean`. In your example, this would be `real, imag = phasor_calibrate(*phasor[1:], *reference_phasor, frequency, lifetime)`. We could just pass through `mean` like in `phasor_filter` with median filter. Although the API feels awkward to require a parameter that is not needed...

I don't think the normalization parameter to `phasor_from_signal` is really needed for the wavelet filter if the intensity is passed to the filter function. Just un- and re-normalize the phasor coordinates with the mean intensity, scaled with the number of samples if absolute intensities are needed.

Anyway, this is developing into a bigger PR with several breaking changes. On the other hand, besides fixing bugs, the API improves too. For example, there is no need any longer for the user to set `skip_axis=0` for multi-harmonic data since that can be detected from the shapes of `mean` and `real`.

cgohlke, 2024-12-04:
Hi @bruno-pannunzio : I propose to rename the existing `phasor_filter` function to `phasor_filter_median` and implement other filters as separate public functions instead of routing additional filters through a single, unified "phasor_filter" interface. The implementation, API, and documentation of a single filter interface will get awkward once more complex filters are added.

Similarly, we should also consider splitting the `phasor_center` function to `phasor_mean` and `phasor_median`...

bruno-pannunzio, 2024-12-04:
> I propose to rename the existing `phasor_filter` function to `phasor_filter_median` and implement other filters as separate public functions instead of routing additional filters through a single, [...]

Hi @cgohlke! I agree with you, this can be a good idea, considering the parameters of the different methods for filtering will be very different and the API may become too crowded, mainly with `phasor_filter`. Please let me know if you want me to make this changes in a PR.

> I don't think the normalization parameter to `phasor_from_signal` is really needed for the wavelet filter if the intensity is passed to the filter function. Just un- and re-normalize the phasor [...]

This is better then! Since most workflows would then use the normalized, this will make it easier than having to normalize again.

> For example, there is no need any longer for the user to set `skip_axis=0` for multi-harmonic data since that can be detected from the shapes of `mean` and `real`.

This is big improvement. Not having to track all this axes for harmonic is excellent.

bruno-pannunzio, 2024-12-20:
Hi @maurosilber! I am trying to integrate pawFLIM into PhasorPy and I have several issues that I believe you can be of great help:

1. Is it strictly necessary to have the information from the first two harmonics? Can this be applied (theoretically) with only one harmonic, or also with more than two?
2. Also, can other harmonics be used instead of the first and second?
3. If I understood correctly, pawFLIM returns the first and second harmonic filtered. What happens with higher harmonics in case there are more than the first two?
4. Are you aware of how pawFLIM handle `NaN` values present in any of the real, imaginary or mean intensity arrays? I tried with some real FLIM examples with `NaN` values and the result depends on the levels specified. After 4 levels, the returned filtered images all contain `NaN` but with lower levels the filtering seems to be working as expected. My guess is that as the level increases, the resolution diminishes and the `NaN` start to interfere and generate `NaN` coefficents. Do you think this can be solved in pawFLIM, or even if it is possible to come up with a solution to ignore `NaN` for filtering including higher levels?

Thanks in advance!

maurosilber, 2024-12-20:
Regarding 1, 2 and 3, it needs the mean intensity (0-th harmonic), the N-th and 2N-th harmonics just to denoise the N-th harmonic. The 0-th and 2N-th are needed to calculate the covariance matrix (of the real and imaginary parts) of a phasor. Hence, you can provide harmonics [0, 1, 2] to denoise the 1st, or [0, 3, 6] to denoise the 3rd.

About point 4, the levels correspond to a maximum averaging area of 2^level. If you have NaNs, they are propagating all over that region. But I think you shouldn't have any NaNs in the Fourier coefficients: no photons or intensity would be 0. Only the phasor ends up with NaNs after dividing the N-th harmonic by the mean intensity (0 / 0).

bruno-pannunzio, 2024-12-20:
Thanks @maurosilber for the very clear response. You made very clear how the harmonic content interacts with the filtering.

Regarding the NaNs, you are right that they should not be present in the Fourier coefficients after transform, if computed directly unnormalized. There are two other reasons why NaNs can be present in the context of PhasorPy:

1. All functions are expected to receive normalized coefficients, thus the wavelet filtering function in PhasorPy should be able to receive normalized coordinates. Then internally the function will unnormalize the data before using pawFLIM and then normalize again after filtering. The main problem (I believe) with this is that mathematically you can't get back the unnormalized coordinates if they are NaN. 

2. Also, thresholding in PhasorPy also assigns thresholded values as NaN, so this can also introduce this values in the coordinates.

Do you think assigning NaN coordinates as (0, 0) a viable option? I am reluctant of this because (0, 0) are valid phasor coordinates, at least in the normalized space. I am not sure they are valid as unnormalized coefficient.

cgohlke, 2024-12-21:
Hi @bruno-pannunzio : I think what Mauro meant was that you can indeed set the Fourier coefficients to zero if G and S are truly NaN and the mean intensity is truly zero.

This is however not possible if the mean intensity is NaN (e.g., the signal may contain NaNs or the threshold filter was applied) or worse, the values are not truthful (keeping in mind that PhasorPy doesn't control the origin of the signals or phasor coordinates). If the wavelet filter algorithm propagates those "irreversible" NaNs over an area of 2^level, so be it, as long as it doesn't crash, is predictable and documented.

closed 2025-05-31 by cgohlke