phasorpy/phasorpy #154: Contribute package to conda-forge

cgohlke, 2024-12-01:
PhasorPy releases are currently published to [PyPI](https://pypi.org/project/phasorpy/) and can be installed with `pip install -U phasorpy` on supported platforms (CPython 3.10-3.13, win_amd64, win_arm64, manylinux2014_x86_64, manylinux2014_aarch64, macosx_11_0_arm64 , macosx_10_9_x86_64).

It would be nice to also support installing PhasorPy directly via the [conda](https://docs.conda.io) package manager. This can be achieved by [contributing a package to conda-forge](https://conda-forge.org/docs/maintainer/adding_pkgs/).

cgohlke, 2025-01-07:
Hi @bruno-pannunzio , are you OK being listed as an additional maintainer for the conda-forge package?

cgohlke, 2025-01-08:
Hi @bruno-pannunzio : it looks like conda-forge has OpenMP on macOS, which is not enabled for the wheels on PyPI. Do you think it's worth enabling OpenMP when building for macOS on conda-forge once https://github.com/conda-forge/staged-recipes/pull/28783 is merged? Do you use conda for development?

bruno-pannunzio, 2025-01-08:
Hi @cgohlke, it can be a good idea to enable OpenMP to allow for future implementations of parallelization on MacOS, which we currently don't support. I don't know how much work that will take, if it is possible, and which priority it has, but enabling can open the road. Although I don't know the implications of enabling it, if it can generate any conflicts, etc. Maybe I can test this before updating the wheels.

I don't really use conda for development. I have used it before for setting up in some simple environments in workshops, etc, but I don't even have it installed. But I can try it if necessary for testing.

cgohlke, 2025-01-08:
I have also added the ptufile library to conda-forge:

- https://github.com/conda-forge/ptufile-feedstock

Two other optional phasorpy dependencies are out of date:

- https://github.com/conda-forge/sdtfile-feedstock
- https://github.com/conda-forge/lfdfiles-feedstock  (no longer maintained?)

cgohlke, 2025-01-08:
> I don't really use conda for development. I have used it before for setting up in some simple environments in workshops, etc, but I don't even have it installed. But I can try it if necessary for testing.

Same for me.

Let's leave out OpenMP on macOS for now since none of us can easily test it. It would require special-casing setup.py for macOS on conda-forge at https://github.com/phasorpy/phasorpy/blob/0bc74c572e4bd528bd009020dcb05d0ac16f3e32/setup.py#L27-L31

In the conda-forge recipe, we would need to add OpenMP to the build requirements for macOS. 

Also, currently conda-forge does not run the test suite.

cgohlke, 2025-01-08:
The [phasorpy-feedstock](https://github.com/conda-forge/phasorpy-feedstock) is now live.

cgohlke, 2025-01-08:
Phasorpy is now available on [Anaconda.org](https://anaconda.org/conda-forge/phasorpy) and can be installed with `conda install conda-forge::phasorpy`.

closed 2025-01-08 by cgohlke