phasorpy/phasorpy #175: Harmonic parameter in `phasor_calibrate` not working as expected

bruno-pannunzio, 2025-01-15:
**Describe the bug**

When the harmonics passed to `phasor_calibrate` function in the `phasor` module include harmonics higher than the size of the first axis of the `real` array, it raises an error of incompatible harmonic although the harmonic is correct.

**To Reproduce**

```Python
from phasorpy.datasets import fetch
from phasorpy.io import read_imspector_tiff
from phasorpy.phasor import phasor_calibrate, phasor_from_signal

harmonics = [1, 2, 4, 8]

signal = read_imspector_tiff(fetch('Embryo.tif'))
frequency = signal.attrs['frequency']
phasor = phasor_from_signal(signal, axis=0, harmonic=harmonics)

reference_signal = read_imspector_tiff(fetch('Fluorescein_Embryo.tif'))
reference_phasor = phasor_from_signal(
    reference_signal, axis=0, harmonic=harmonics
)

real, imag = phasor_calibrate(
    *phasor[1:],
    *reference_phasor,
    frequency=frequency,
    lifetime=4.2,
    harmonic=harmonics,
)
```

A Python traceback if available, for example:

```Python traceback
Traceback (most recent call last):
File ~/phasorpy/src/phasorpy/_utils.py:389, in parse_harmonic(harmonic, harmonic_max)
IndexError: harmonic=[1, 2, 4, 8] element > 4]
```

Information how PhasorPy was installed (pip, conda, or other) and the console output of:

```
$ python -m phasorpy versions
Python-3.13.1
phasorpy-0.4.dev
numpy-2.2.1
tifffile-2025.1.10
imagecodecs-n/a
lfdfiles-2024.10.24
sdtfile-2024.12.6
ptufile-2025.1.13
matplotlib-3.10.0
scipy-1.15.1
skimage-n/a
sklearn-n/a
pandas-2.2.3
xarray-2025.1.1
click-8.1.8
pooch-v1.8.2
```

bruno-pannunzio, 2025-01-15:
Hi @cgohlke, reviewing some workflow for the implementation of the wavelet filter, I came upon a bug related to multi-harmonic calibration (again).

I know the issue raises because of this section of the code in `phasor_calibrate`:

```Python
has_harmonic_axis = reference_mean.ndim + 1 == reference_real.ndim
harmonic, _ = parse_harmonic(
   harmonic, reference_real.shape[0] if has_harmonic_axis else None
)
```

The error occurs because it is assumed that the max harmonic corresponds to the size of the first dimension of `reference_real`, which in the example above is not the case. We should change how to handle getting the max harmonic number in this case to avoid this to happen, while also being able to support the `all` option for the `harmonic` parameter. What do you think? Should I open a PR to discuss this in more detail with potential solutions?

bruno-pannunzio, 2025-01-15:
Changing the selected portion of the code in `phasor_calibrate` to the following solves the issue (at least in this case):

```Python
has_harmonic_axis = reference_mean.ndim + 1 == reference_real.ndim
harmonic, _ = parse_harmonic(
    harmonic, reference_real.shape[0] if has_harmonic_axis and harmonic == 'all' else None
)
```

The problem is that in the case of `harmonic = 'all'` it assumes `harmonic` contains harmonics from 0 to `range(reference_real.shape[0]`. Is this correct?

cgohlke, 2025-01-15:
Obviously the tests are incomplete. The bug should be easy to fix.

cgohlke, 2025-01-15:
> in the case of harmonic = 'all' it assumes harmonic contains harmonics from 0 to range(reference_real.shape[0]

Yes, according to the parameter documentation.

closed 2025-01-16 by bruno-pannunzio