phasorpy/phasorpy #200: Implementation of cursor segmentation using machine learning

R-Blanco802, 2025-03-05:
It was observed that when a user of the PhasorPy library has obtained the phasors and wants to pseudo-color the image, they will need to manually analyze the distribution of the points in the phasor space in order to select which ROI will be coloured.
This may become cumbersome if they have a large number of images to analyze, and may also introduce biases in users. For example, a user who is trained on certain type of phasor images may tend to extract similar observations in the future, even when analyzing differently distributed phasor points which could lead to misinterpretation.

This issue proposes to automatize the observation of the distribution of phasor points, and selecting the cursors (further corrections are placed on the user, and how does the pseudo-coloured image looks like to them specifically).

In order to do this, I suggest using some scikit-learn (sklearn) modules such as the GMM (Gaussian Mixture Models) or k-means to find clusters in the phasor image and use them to choose centers, radii (in case of elliptical cursors), etc. for the cursors.

I am preparing a Pull Request for this, but I wanted your input on how the module's functions should operate and how to address certain problems: 
- One challenge is handling NaN values, as they pose a problem for sklearn. This was temporarily mitigated by replacing NaNs with zeros. 
- Another complication arises when two cursors overlap in the same region causing images to loose colour. For instance, if a blue and a red cursor are present, the expected result would be a mix of both with purple in the middle. However, the image appears predominantly gray with some blue, red and purple.

Thanks in advance!

bruno-pannunzio, 2025-03-05:
Hi @R-Blanco802! Thanks for showing interest in the PhasorPy project and for contributing to it. The automatic clustering is something we are very interested in having in the library, so this is much welcome.

> * One challenge is handling NaN values, as they pose a problem for sklearn. This was temporarily mitigated by replacing NaNs with zeros.

The `NaN` handling is something we are aiming for the library, since many functions return `NaN` for invalid values (such as an invalid fourier transformation or after thresholding). This can be a problem, particularly when using other libraries as 'sklearn'. The problem with replacing `NaN` with zeroes is that this are valid values for phasor coordinates and can affect the analysis if not handled properly. Maybe when you open the PR, we can discuss based on your implementation what can be an efficient way to handle the `NaN` without skewing the analysis. This can be done through masks or maybe other alternatives.

> * Another complication arises when two cursors overlap in the same region causing images to loose colour. For instance, if a blue and a red cursor are present, the expected result would be a mix of [...]

This seems to be a problem related to the cursors itself, maybe of the `pseudo_color` or the `mask_from_*` functions. Either way, I believe this can be raised in a separate issue with some code example to replicate the problem.

cgohlke, 2025-03-06:
> Another complication arises when two cursors overlap in the same region causing images to loose colour. For instance, if a blue and a red cursor are present, the expected result would be a mix of both with purple in the middle. 

That can be achieved with screen blending, which is implemented but not available in the `pseudo_color` function. You could try to experiment with the `_blend_screen` mode at the following lines:

https://github.com/phasorpy/phasorpy/blob/8fd32048b52586c0e86db3655c40e27c5c320fd4/src/phasorpy/cursors.py#L495-L499

Frankly, the whole cursors module is half-baked. There was some discussion at the beginning of this project how to handle, for example, overlapping and non-binary cursor masks (like produced by GMM), but no one stepped up and the module was left as is.

cgohlke, 2025-05-31:
Closed by #201

closed 2025-05-31 by cgohlke