phasorpy/phasorpy #206: # Reorganize package namespace

cgohlke, 2025-03-20:
The purpose of this issue is to discuss if or how to reorganize the phasorpy package namespace, internal organization, and documentation before the next release (which should be the last release with large breaking changes).

The main arguments for reorganizing the package are that many imports feel clumsy (for example, `from phasorpy.phasor import phasor_from_signal`) and some modules grew too large (for example, `phasor.py` and `io.py`).

I propose the following reorganization:

1. Split the `phasor.py` module into `phasor`, `lifetime`, `fret`, and `filter`.

2. Export all public attributes from the `phasor`, `lifetime`, `fret`, `filter`, `components` `cursors`, `cluster`, and `color` modules from the `phasorpy` namespace. Above import will become `from phasorpy import phasor_from_signal`. 

3. Make the `phasor`, `lifetime`, `fret`, `filter`, `components` `cursors`, and `cluster` modules private, for example, prefix the names with an underscore. Should the `color` module be made private too?

4. Move the `io` module into a directory and split the implementation into separate, private files as necessary. All functions will be exported from the `phasorpy.io` namespace. The reasons for not moving io functions to the main namespace are that the `io` namespace is well established among other packages and functions are found faster.

5. Leave the `plot` module as is for now. It could be later moved into a directory and split up. The reason for not moving functions and classes from the plot module into the main namespace is that they are specific to the matplotlib plotting library.

6. Leave the `cli`, `datasets`, `utils`, and `version` modules as is. These are not directly related to PhasorPy's core functionality.

7. Rename or move some functions to make them consistent in the global namespace:

      - `n_fractions_from_phasor` -> `phasor_component_?`
      - `two_fractions_from_phasor` -> `phasor_component_?`
      - `graphical_component_analysis` -> `phasor_component_graphical`
      - `mask_from_circular_cursor` -> `phasor_mask_circular`
      - `mask_from_elliptic_cursor` -> `phasor_mask_elliptic`
      - `mask_from_polar_cursor` -> `phasor_mask_polar`
      - `cursors.pseudo_color` -> `color.pseudo_color` or `image_overlay` or `image_blend`
      - `CATEGORICAL` -> ?
      - `SRGB_SPECTRUM` -> ?
      - `float2int` -> ?
      - `wavelength2rgb` -> ?

8. Reorganize tests reflecting the new package structure.

9. Reorganize documentation to document individual functions and classes instead of modules.

10. Update tutorials.

The proposed changes are largely incompatible and will break

- existing code
- documentation links
- git history

The main proposed change is moving most functions into a flat namespace. The `phasorpy` namespace will contain about 42 functions and constants. If the library is expected to grow much, the size of the namespace might become an issue (for example import time, tab completion).

cgohlke, 2025-03-20:
> The phasorpy namespace will contain about 42 functions and constants.

- `CATEGORICAL`
- `SRGB_SPECTRUM`
- `__version__`
- `float2int`
- `graphical_component_analysis`
- `lifetime_fraction_from_amplitude`
- `lifetime_fraction_to_amplitude`
- `lifetime_from_frequency`
- `lifetime_to_frequency`
- `lifetime_to_signal`
- `phasor_at_harmonic`
- `phasor_calibrate`
- `phasor_center`
- `phasor_divide`
- `phasor_filter_median`
- `phasor_filter_pawflim`
- `phasor_from_apparent_lifetime`
- `phasor_from_fret_acceptor`
- `phasor_from_fret_donor`
- `phasor_from_lifetime`
- `phasor_from_polar`
- `phasor_from_signal`
- `phasor_mask_circular`
- `phasor_mask_elliptic`
- `phasor_mask_polar`
- `phasor_multiply`
- `phasor_normalize`
- `phasor_semicircle`
- `phasor_threshold`
- `phasor_to_apparent_lifetime`
- `phasor_to_complex`
- `phasor_to_polar`
- `phasor_to_principal_plane`
- `phasor_to_signal`
- `phasor_transform`
- `polar_from_apparent_lifetime`
- `polar_from_reference`
- `polar_from_reference_phasor`
- `polar_to_apparent_lifetime`
- `pseudo_color`
- `spectral_vector_denoise`
- `two_fractions_from_phasor`
- `wavelength2rgb`

cgohlke, 2025-03-20:
> the size of the namespace might become an issue

For comparison, `len(numpy.__all__) == 499`

bruno-pannunzio, 2025-03-21:
Hi @cgohlke! I think all of the ideas 1-6 are excellent. I totally agree with what is proposed and the splitting of the phasor module into the other 4 modules.

Regarding the renaming of some functions, I think this is a larger discussion that is better to have separately but since there are so many functions starting with `phasor_*`, sometimes looking for them is a bit cumbersome. Perhaps we could keep that in the functions that have `to` or `from` and the others remove since most are intended for phasors. Just in the ones (for example the ones that work with polar coordinates) specify that in the name.

I don't know, it's just an idea that maybe is not a good one. I also like the naming structures as it is because is very informative. I bring this just for discussion.

cgohlke, 2025-03-21:
> many functions starting with phasor_*, sometimes looking for them is a bit cumbersome

I agree. I was hoping that tab completion could help. I think the prefixes are necessary in a flat namespace and it's better to be explicit. Many of the simplified function names (`multiply`, `normalize`, `transform`) could be confused with different functionality. An option is to move those functions into the `phasor` namespace and drop the prefix, like `from phasorpy import phasor;phasor.from_signal(...)`. Then do the same with `lifetime`, `polar`, etc. For example, `from phasorpy import polar;polar.from_apparent_lifetime(...)`. But then users would expect `polar.to_phasor()`, which is actually `phasor.from_polar()`. We could provide simple convenience functions, but duplicating much functionality doesn't feel quite right in a library and the functions still need to be documented, tested, and maintained. If you have time, try to propose a package layout with sub-namespaces and post it here.

cgohlke, 2025-05-31:
Stale

PR #245 (2025-06-17): Reorganize plot module
## Description

This PR reorganizes the `plot` module by moving the implementation from `plot.py` into several private submodules in a `plot` folder. The tests are reorganized accordingly. The public API remains unchanged.

Partially fixes #206.

## Checklist

- [ ] The pull request title and description are concise.
- [ ] Related issues are linked in the description.
- [ ] New dependencies are explained.
- [ ] The source code and documentation can be distributed under the [MIT license](https://www.phasorpy.org/docs/stable/license/).
- [ ] The source code adheres to [cod [... cut]
closed 2025-06-04 by cgohlke