phasorpy/phasorpy #212: Colour selection of contour plot

stllgrnr, 2025-03-27:
Hi everyone,
I have a pretty specific question:
I measured multiple FLIM images of the same position (using different detection windows) and would like to plot one contour line of each in a phasor plot (see image) to compare the shift of the phasor over the detection wavelengths. 
[image]
I created this image using 26 colourmaps with only one colour and phasorplot.contour(). As far as I understood, matplotlib.pyplot.contour() also accepts a colour as input - as long as no colourmap is specified. The problem is that with no colourmap specified, phasorplot.contour() sets cmap='Blues' which causes a ValueError('Either colors or cmap must be None')  if I specify colours.
Is it possible to adapt this so one can input a colour for the contour plot?
If not - no problem, I found a workaround but wanted to say thank you for working on this cool project!
Best,
Stella

cgohlke, 2025-03-27:
If I understood this issue correctly, the `cmap` parameter should be ignored if `colors` is set, which is the default behavior in `matplotlib.pyplot.contour`. Please confirm and I will create a PR for that.

closed 2025-03-27 by cgohlke

cgohlke, 2025-03-27:
Hi @stllgrnr : this should be fixed in #213 , which is now in the development version.

PR #213 (2025-03-27): Ignore PhasorPlot.contour cmap if colors is set
## Description

This PR fixes `ValueError: Either colors or cmap must be None` if `PhasorPlot.contour` is called with a `colors` parameter.

Closes #212.

## Checklist

- [ ] The pull request title and description are concise.
- [ ] Related issues are linked in the description.
- [ ] New dependencies are explained.
- [ ] The source code and documentation can be distributed under the [MIT license](https://www.phasorpy.org/docs/stable/license/).
- [ ] The source code adheres to [code standards](https://www.phasorpy.org/docs/stable/contributing/#cod [... cut]