phasorpy/phasorpy #246: Reference Requirements

ngoxlzh, 2025-06-19:
Hello there, 

I've got some followup questions regarding issue #205 . 
Because of our setup, it seems we are not able to measure at half gain. I was not able to completely get rid of the cutoff of the decay curve (zeroed entries). Since I mostly work with short lifetime samples (bound/free NADH), this hopefully won't have the greatest impact. 

In the meantime I was able re-image my samples and obtain some rhodamine G6 and fluorescein to use as references. I expect the following lifetimes: 

NADH (free) 0.4ns, peak at 370nm
NADH (bound) 1.2ns, peak at 370nm
rhodamine G6 4ns, peak at 525nm
fluorescein 4ns, peak at 500nm

Our turn-key laser is pretty old at this point and at different wavelengths it has a different beam shape. This is why I suspect my rhodamine/fluorescein reference measurements at longer wavelengths (~950nm) may not apply to my NADH measurements at 740nm. Additionally, the lasers power varies about 1W between the longer and shorter wavelengths (Max power ~3W at 800nm). We do not have a way to accurately measure the power that arrives at the sample. 

In order to ensure the beam shape stayed the same between signal and reference, I opted to measure rhodamin at 740nm. I did have to increase the laser power though to get a similar amount of signal. 

What is seems curious to me is that according to my evaluations with SPCImage, the lifetime of my rhodamine sample was only about 1ns. I found a paper saying that rhodamines lifetime may be shorter at higher temperatures  so maybe the laser heats the sample up. It may also be that the commonly known lifetime (4ns) only applies to the main peak of the fluorescence spectrum. 

Now to get to PhasorPy. To evaluate a NADH signal that I imaged right after the rodamine, I tried both 4ns and 1ns in phasor_calibrate. The result from using 1ns seems to make more sense. However, once again most of the pixels are outside the semicircle. I did make sure to cut off the zeroed entries, as you suggested in #205. 
Do you have some idea what should be done differently?

This is my new dataset: https://lzh-cloud.solidstore.de/index.php/s/AXFsjAqkHDA7Seq

Could you please specify which system settings/experiment conditions need to stay the same? I suspect this might be an issue. 

Best regards

ngoxlzh, 2025-06-19:
Sorry about that, I mixed up a lot here. I uploaded some new files to the link. The ones with the date 20250520 are an NADH sample and a rhodamine and fluorescein one, but using different exitation wavelengths and some different filters before the detectors (490-575nm for fluorescein/rhodamine and 420-520nm for NADH). The rhodamine sample from 20250611 was imaged using the same wavelength and filter as the NADH one from 20250520.
Please disregard the first channel, relevant data is only in the second one. 
I uploaded the python file as well. It's pretty much just your code suggestion from the previous issue. 

For data, I used the rhodamine sample from 20250611 and the NADH one from 20250520. I do get a nice, seemingly monoexponential result for the NADH now, but with a lifetime of 0.5ns and some strange "tail" towards the outside. 

[image]

cgohlke, 2025-06-20:
> I uploaded some new files to the link

```
Share not found

This share does not exist or is no longer available
```

ngoxlzh, 2025-06-26:
Thank you for the feedback. I'll be able to do some imaging next week. 

I'd still like to have a list of what parameters and experiment conditions need to stay the same between the refernce and sample image. What I've gatherd: 

- most system parameters
- matching signal intensity
- matching beam shape
- same filters in beamline
- "enough" signal

I would appreciate, If you could add to this list.

ngoxlzh, 2025-07-02:
Ok so I reimaged my samples/reference at a similar intensity and was able to get a pretty decent result. I made sure to use the same wavelength, laser power and record until I had about the same signal/referene intensity. Thank you for your help. 

[image]

I would still appreciate If you could add to my list from the previous comment. 
Thank you for your time

cgohlke, 2025-07-02:
> list of what parameters and experiment conditions need to stay the same between the refernce and sample image. What I've gatherd:
>
>   - most system parameters
>   - matching signal intensity
>   - [...]

Some other requirements were already mentioned, others are general or specific to instrumentation/method. From memory:

- no background, or background removed
- no cutoffs, the whole decay must be covered by the laser period
- no photon pileup
- no change in laser/clock synchronization
- same IRF and ideally enough samples to resolve it

Some of those could be handled specifically but it's probably better to avoid them in the first place. 

@bruno-pannunzio : anything else to add?

bruno-pannunzio, 2025-07-03:
Hi @ngoxlzh and @cgohlke.

Christoph put together a very thorough and thoughtful list of requirements, and he raised some important points about the signal behavior. I wasn’t able to download the files, I believe the link may have expired.

If there’s a bump in the signal, that could explain the incorrect phasor. We've encountered similar issues with files from different systems, and it can be quite problematic. Such bumps can result from pile-up effects or internal laser reflections. In the case of reflections, reducing the laser power might help. Just make sure it's consistent for both the sample and the reference acquisitions.

As for the list of conditions, I agree that all factors affecting the electronic signal processing should remain constant. That consistency is precisely why calibration is necessary in the first place.

cgohlke, 2025-07-03:
> If there’s a bump in the signal

It was a slight bump, nothing like we've seen in Leica data. Nevertheless, reducing/matching the intensities seems to have resolved the issue.

closed 2025-07-03 by cgohlke